bluerobotics/bluerobotics.github.io #6: T200 Advance ratio

Txon, 2016-03-10:
Hi friends:

I'm into a design that will use your T200 thruster. Albeit you've provided a lot of information, I'd like to know a few details about the testing conditions related to the published curves.

May I know which "free flow speed" was used for the Thrust curves plotting ? Or are those curves obtained in static conditions ?

I need that data in order to achieve some hidrodynamic goals.

Thanks in advance and ... kind regards

rjehangir, 2016-05-23:
Hi Txon, sorry we never answered this! The curves on our documentation were all collected at near-static conditions. We don't have any data for thrust with non-zero flow speed.

Best, Rusty

Txon, 2016-05-25:
Hi Rustom, better late thern never. Thanks for your answer. I'll try to
perform some calculations for non-zero free stream speed.

Thanks again.

closed 2019-03-19 by jaxxzer