bluerobotics/bluerobotics.github.io #8: Inconsistent documentation for thruster config

binary42, 2016-06-07:
Inconsistent documentation for thruster config. 

In the build instructions for the Bluerov, the thruster order is not the same as the thruster numbering on the ArduSub documentation site. Thrusters 1,2 and 3,4 are mismatched.

rjehangir, 2016-06-07:
@binary42 Good catch! We changed the numbering scheme in ArduSub recently but have updated those diagrams yet. We'll do that right away.

binary42, 2016-06-07:
Excellent. I found it when trying to configure the Navio+ with the kit...it had not been without great frustration. I am preparing a write for the forums about it. Very discouraging. I think there is a bug in the navio build of ArduSub. The Navio+ is even flaky with ArduCopter when I test it with a quad. Anyway, that's for later a different post. I've picked up a Pixhawk in any event. Once I get this all straightened out, I think it would be a good idea to create Docker images for the companion computers. I think it will save people days of anger and frustration in setup.

rjehangir, 2016-06-07:
@binary42 Okay. I had trouble getting it working well on the Navio2 as well. The linux versions seem to be very "beta" still.

binary42, 2016-06-25:
Documentation is now correct. Thanks!

closed 2016-06-25 by binary42