bluerobotics/bluerobotics.github.io #30: BROV2 kit assembly page difficult to view

jaxxzer, 2017-07-19:
The text on the page bounces around for a couple of minutes as all of the many images load.

We should compress/resize the images so they load faster, or have the image containers initialize to the correct height before the image is loaded.

rjehangir, 2017-07-19:
Good idea. There are a few ways to do this I think:

- Add javascript to put each `img` tag into a `div` with set height
- Compress the images, probably changing them to jpegs would make a huge difference
- Add a "lazy load" javascript library so that pictures only load when you scroll to that part of the page but are properly sized already, which would minimize loading time.

jaxxzer, 2019-04-02:
This is much better behaved for me today, due to faster response from the server. I'm not sure how it will look on a low bandwidth connection however, and if something can still be done there in terms of not bouncing text around while things load.

jaxxzer, 2019-04-02:
I think the effort will outweigh the benefit at this point, so will close in that spirit.

closed 2019-04-02 by jaxxzer